madhephaestus/ESP32AnalogRead #1: Robustness and minor optimization

RobTillaart, 2020-04-12:
The code will give a problem if the user calls readVoltage() before the call of attach().

solution 1: move pin and channel assignment to constructor.
solution 2: have a boolean flag that indicated that the object is properly attached/ assigned/ configured

Furthermore a minor optimization (multiplication tends to be slightly faster on most processors)
```
float ESP32AnalogRead::readVoltage() { return readMiliVolts() * 0.001; };
```

madhephaestus, 2020-04-12:
I just recompiled master and checked the binaries, no change. 

I also removed liblockfile-dev and recompiled, so there is no possible way the dep is being used. The compile worked, and the binaries are identical. AFAIK, RXTX had always used lock files, but had not used a lockfile library. We added the library, but this caused compile issues across the platforms. The failover when the lockfile library is not availible, it uses the old built in lock file system. Is this the behavior your seeing?

closed 2020-04-12 by madhephaestus

madhephaestus, 2020-04-12:
I added your suggestions.

commit 9b97cc5a (2020-04-12): Close #1