madhephaestus/ESP32AnalogRead #2: Need your help, again

tombo9999, 2020-10-03:
Hi Kevin, sorry to bother you, again. Hope you can help me. Thanks in advance, Luca
I'm using pin 33 on Ai-Thinker ESP32-Cam (soldered to red internal led to have one more pin).
My board ADC VRef calibration = 1093mV (efuse)
I just want to check a 3.7V Lipo battery connected to pin 33 with voltage divider (No 2 100ohm resistors).
At first I tried "adc1_example_main.c" with ESP-IDF from Espressif and increasing NO_OF_SAMPLES to 256 I can get quite stable readings: 42 exact value over a total of 86 ( with difference from -0.01 to 0.01).
But I wanted to use Arduino and I found your library wich give me 9 exact value over a total of 86.
One more thing:
- in esp-idf I need to multiply by the constant 1.967
- with arduino the constant it's 2.015 but the readings are less stable (difference from -0.13 to 0.10)
```
/* ESP32AnalogReadTest */

#include "Arduino.h"
#include <ESP32AnalogRead.h>
ESP32AnalogRead adc;

void setup()
{
	adc.attach(33);
	Serial.begin(115200);
}

void loop()
{
	delay(1000);
	Serial.println("Voltage = "+String(adc.readVoltage()*2.015));
}
```
```
/* ADC1 Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "driver/adc.h"
#if CONFIG_IDF_TARGET_ESP32
#include "esp_adc_cal.h"
#endif

#define DEFAULT_VREF    1100        //Use adc2_vref_to_gpio() to obtain a better estimate
// ADC VRef calibration: 1093mV
#define NO_OF_SAMPLES   64          //Multisampling
// #define NO_OF_SAMPLES  256          //Multisampling

#if CONFIG_IDF_TARGET_ESP32
static esp_adc_cal_characteristics_t *adc_chars;
// static const adc_channel_t channel = ADC_CHANNEL_6;     // GPIO33 LD //GPIO34 if ADC1, GPIO14 if ADC2
static const adc_channel_t channel = ADC_CHANNEL_5;     // GPIO33 Luca
static const adc_bits_width_t width = ADC_WIDTH_BIT_12;
#elif CONFIG_IDF_TARGET_ESP32S2
static const adc_channel_t channel = ADC_CHANNEL_6;     // GPIO7 if ADC1, GPIO17 if ADC2
static const adc_bits_width_t width = ADC_WIDTH_BIT_13;
#endif
// static const adc_atten_t atten = ADC_ATTEN_DB_0;
static const adc_atten_t atten = ADC_ATTEN_DB_11;
static const adc_unit_t unit = ADC_UNIT_1;

#if CONFIG_IDF_TARGET_ESP32
static void check_efuse(void)
{
    //Check TP is burned into eFuse
    if (esp_adc_cal_check_efuse(ESP_ADC_CAL_VAL_EFUSE_TP) == ESP_OK) {
        printf("eFuse Two Point: Supported\n");
    } else {
        printf("eFuse Two Point: NOT supported\n");
    }

    //Check Vref is burned into eFuse
    if (esp_adc_cal_check_efuse(ESP_ADC_CAL_VAL_EFUSE_VREF) == ESP_OK) {
        printf("eFuse Vref: Supported\n");
    } else {
        printf("eFuse Vref: NOT supported\n");
    }
}

static void print_char_val_type(esp_adc_cal_value_t val_type)
{
    if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP) {
        printf("Characterized using Two Point Value\n");
    } else if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF) {
        printf("Characterized using eFuse Vref\n");
    } else {
        printf("Characterized using Default Vref\n");
    }
}
#endif

void app_main(void)
{
#if CONFIG_IDF_TARGET_ESP32
    //Check if Two Point or Vref are burned into eFuse
    check_efuse();
#endif

    //Configure ADC
    if (unit == ADC_UNIT_1) {
        adc1_config_width(width);
        adc1_config_channel_atten(channel, atten);
    } else {
        adc2_config_channel_atten((adc2_channel_t)channel, atten);
    }

#if CONFIG_IDF_TARGET_ESP32
    //Characterize ADC
    adc_chars = calloc(1, sizeof(esp_adc_cal_characteristics_t));
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize(unit, atten, width, DEFAULT_VREF, adc_chars);
    print_char_val_type(val_type);
#endif

    //Continuously sample ADC1
    while (1) {
        uint32_t adc_reading = 0;
        //Multisampling
        for (int i = 0; i < NO_OF_SAMPLES; i++) {
            if (unit == ADC_UNIT_1) {
                adc_reading += adc1_get_raw((adc1_channel_t)channel);
            } else {
                int raw;
                adc2_get_raw((adc2_channel_t)channel, width, &raw);
                adc_reading += raw;
            }
        }
        adc_reading /= NO_OF_SAMPLES;
#if CONFIG_IDF_TARGET_ESP32
        //Convert adc_reading to voltage in mV
        uint32_t voltage = esp_adc_cal_raw_to_voltage(adc_reading, adc_chars);
		// float voltage1 = (voltage*1.977 +0.5)/1000;
		// float voltage1 = (voltage*1.969 -0.005)/1000;
		float voltage1 = (voltage*1.967)/1000;
		// #if 
        printf("Raw: %d\tVoltage: %.2f V\n", adc_reading, voltage1); // Luca
		printf("Raw: %d\tVoltage: %dmV\n", adc_reading, voltage);
#elif CONFIG_IDF_TARGET_ESP32S2
        printf("ADC%d CH%d Raw: %d\t\n", unit, channel, adc_reading);
#endif
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```

madhephaestus, 2020-10-05:
the ADC on the esp32 is wierd. It has no ability to read voltages between 0 and 0.13v not in any mode. Are your errors in that range?

tombo9999, 2020-10-11:
Thanks Kevin!
My errors are in the range -0.01 and 0.01V. Yes I know the ADC on the ESP32 is weird.
Found a workaround for my problem, using down-sampling the result are quite stable.
At least now can check a 3.7V Lipo battery and the reading is quite correct compared with my voltmeter.

closed 2020-10-11 by tombo9999