madhephaestus/ESP32AnalogRead #8: Support for ESP32-C3 ?

Konstantin-G, 2022-05-18:
Hi, 
great library! thanks.
Are you thinking about adding Support for ESP32-C3 ? 
It will be awesome!
Thanks
Konstantin

madhephaestus, 2022-05-19:
What specifically is the issue? Have you tried it yet?

Konstantin-G, 2022-05-19:
think problem is ESP32-C3 has less pins (ADC pins too) i’m getting errors like: 
'ADC1_CHANNEL_5' was not declared in this scope chan = ADC1_CHANNEL_5; 
for each pin C3 does not have.
The whole error message here:

Arduino: 1.8.19 (Mac OS X), Board: "ESP32C3 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 115200, None"

/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp: In member function 'uint32_t ESP32AnalogRead::readMiliVolts()':
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:82:11: error: 'ADC1_CHANNEL_5' was not declared in this scope
    chan = ADC1_CHANNEL_5;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:82:11: note: suggested alternative: 'ADC1_CHANNEL_4'
    chan = ADC1_CHANNEL_5;
           ^~~~~~~~~~~~~~
           ADC1_CHANNEL_4
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:85:11: error: 'ADC1_CHANNEL_6' was not declared in this scope
    chan = ADC1_CHANNEL_6;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:85:11: note: suggested alternative: 'ADC1_CHANNEL_4'
    chan = ADC1_CHANNEL_6;
           ^~~~~~~~~~~~~~
           ADC1_CHANNEL_4
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:88:11: error: 'ADC1_CHANNEL_7' was not declared in this scope
    chan = ADC1_CHANNEL_7;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:88:11: note: suggested alternative: 'ADC1_CHANNEL_4'
    chan = ADC1_CHANNEL_7;
           ^~~~~~~~~~~~~~
           ADC1_CHANNEL_4
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:101:11: error: 'ADC2_CHANNEL_1' was not declared in this scope
    chan = ADC2_CHANNEL_1;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:101:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_1;
           ^~~~~~~~~~~~~~
           ADC2_CHANNEL_0
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:104:11: error: 'ADC2_CHANNEL_2' was not declared in this scope
    chan = ADC2_CHANNEL_2;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:104:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_2;
           ^~~~~~~~~~~~~~
           ADC2_CHANNEL_0
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:107:11: error: 'ADC2_CHANNEL_3' was not declared in this scope
    chan = ADC2_CHANNEL_3;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:107:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_3;
           ^~~~~~~~~~~~~~
           ADC2_CHANNEL_0
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:110:11: error: 'ADC2_CHANNEL_4' was not declared in this scope
    chan = ADC2_CHANNEL_4;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:110:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_4;
           ^~~~~~~~~~~~~~
           ADC2_CHANNEL_0
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:113:11: error: 'ADC2_CHANNEL_5' was not declared in this scope
    chan = ADC2_CHANNEL_5;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:113:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_5;
           ^~~~~~~~~~~~~~
           ADC2_CHANNEL_0
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:116:11: error: 'ADC2_CHANNEL_6' was not declared in this scope
    chan = ADC2_CHANNEL_6;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:116:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_6;
           ^~~~~~~~~~~~~~
[... 8 lines cut ...]
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:122:11: error: 'ADC2_CHANNEL_8' was not declared in this scope
    chan = ADC2_CHANNEL_8;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:122:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_8;
           ^~~~~~~~~~~~~~
           ADC2_CHANNEL_0
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:125:11: error: 'ADC2_CHANNEL_9' was not declared in this scope
    chan = ADC2_CHANNEL_9;
           ^~~~~~~~~~~~~~
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:125:11: note: suggested alternative: 'ADC2_CHANNEL_0'
    chan = ADC2_CHANNEL_9;
           ^~~~~~~~~~~~~~
           ADC2_CHANNEL_0
/Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:152:47: error: invalid conversion from 'int32_t*' {aka 'long int*'} to 'int*' [-fpermissive]
   adc2_get_raw(adc2_channel, ADC_WIDTH_12Bit, &raw);
                                               ^~~~
[... cut ...]
                 from /Users/konstantinlaska/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:9:
/Users/konstantinlaska/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3-RC1/tools/sdk/esp32c3/include/driver/include/driver/adc.h:381:81: note:   initializing argument 3 of 'esp_err_t adc2_get_raw(adc2_channel_t, adc_bits_width_t, int*)'
 esp_err_t adc2_get_raw(adc2_channel_t channel, adc_bits_width_t width_bit, int *raw_out);
                                                                            ~~~~~^~~~~~~
exit status 1
Error compiling for board ESP32C3 Dev Module.

Konstantin-G, 2022-05-24:
any thoughts?

Konstantin-G, 2022-05-27:
I made it works, pls give me write rights, I will add it to new branch.

madhephaestus, 2022-05-27:
Awesome!! The way git hub works is that you fork the repo, make changes in
your fork, then click the submit pull request button in your fork. That
will give us a chance to review the changes and merge them. I am looking
forward to supporting the C3 core!

You rock!

closed 2022-06-06 by madhephaestus