madhephaestus/ESP32AnalogRead #13: Update kernel esp32 and error library

ybpvin, 2024-06-15:
Hi!
ESP32AnalogRead library error occurred after kernel update. 0.3.0.
Maybe there is a solution to this problem?

Arduino IDE 2.3.2. for Mac os
kernel  https://github.com/espressif/arduino-esp32
ESP32AnalogRead. 0.3.0

compiler message aduino

/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp: In member function 'uint32_t ESP32AnalogRead::readMiliVolts()':
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:187:35: error: 'ADC_WIDTH_12Bit' was not declared in this scope; did you mean 'ADC_WIDTH_MAX'?
  187 |                 adc1_config_width(ADC_WIDTH_12Bit);
      |                                   ^~~~~~~~~~~~~~~
      |                                   ADC_WIDTH_MAX
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:187:17: error: 'adc1_config_width' was not declared in this scope
  187 |                 adc1_config_width(ADC_WIDTH_12Bit);
      |                 ^~~~~~~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:219:49: error: 'ADC_ATTEN_11db' was not declared in this scope; did you mean 'ADC_ATTEN_DB_6'?
  219 |                 adc1_config_channel_atten(chan, ADC_ATTEN_11db);
      |                                                 ^~~~~~~~~~~~~~
      |                                                 ADC_ATTEN_DB_6
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:219:17: error: 'adc1_config_channel_atten' was not declared in this scope
  219 |                 adc1_config_channel_atten(chan, ADC_ATTEN_11db);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:259:49: error: 'ADC_ATTEN_11db' was not declared in this scope; did you mean 'ADC_ATTEN_DB_6'?
  259 |                 adc2_config_channel_atten(chan, ADC_ATTEN_11db);
      |                                                 ^~~~~~~~~~~~~~
      |                                                 ADC_ATTEN_DB_6
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:259:17: error: 'adc2_config_channel_atten' was not declared in this scope
  259 |                 adc2_config_channel_atten(chan, ADC_ATTEN_11db);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:281:23: error: 'adc1_get_raw' was not declared in this scope
  281 |                 raw = adc1_get_raw(adc1_channel);
      |                       ^~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:285:44: error: 'ADC_WIDTH_12Bit' was not declared in this scope; did you mean 'ADC_WIDTH_MAX'?
  285 |                 adc2_get_raw(adc2_channel, ADC_WIDTH_12Bit, &raw);
      |                                            ^~~~~~~~~~~~~~~
      |                                            ADC_WIDTH_MAX
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:285:17: error: 'adc2_get_raw' was not declared in this scope
  285 |                 adc2_get_raw(adc2_channel, ADC_WIDTH_12Bit, &raw);
      |                 ^~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp: In member function 'uint16_t ESP32AnalogRead::readRaw()':
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:491:35: error: 'ADC_WIDTH_12Bit' was not declared in this scope; did you mean 'ADC_WIDTH_MAX'?
  491 |                 adc1_config_width(ADC_WIDTH_12Bit);
      |                                   ^~~~~~~~~~~~~~~
      |                                   ADC_WIDTH_MAX
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:491:17: error: 'adc1_config_width' was not declared in this scope
  491 |                 adc1_config_width(ADC_WIDTH_12Bit);
      |                 ^~~~~~~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:523:49: error: 'ADC_ATTEN_11db' was not declared in this scope; did you mean 'ADC_ATTEN_DB_6'?
  523 |                 adc1_config_channel_atten(chan, ADC_ATTEN_11db);
      |                                                 ^~~~~~~~~~~~~~
      |                                                 ADC_ATTEN_DB_6
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:523:17: error: 'adc1_config_channel_atten' was not declared in this scope
  523 |                 adc1_config_channel_atten(chan, ADC_ATTEN_11db);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:563:49: error: 'ADC_ATTEN_11db' was not declared in this scope; did you mean 'ADC_ATTEN_DB_6'?
  563 |                 adc2_config_channel_atten(chan, ADC_ATTEN_11db);
      |                                                 ^~~~~~~~~~~~~~
      |                                                 ADC_ATTEN_DB_6
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:563:17: error: 'adc2_config_channel_atten' was not declared in this scope
  563 |                 adc2_config_channel_atten(chan, ADC_ATTEN_11db);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:584:23: error: 'adc1_get_raw' was not declared in this scope
  584 |                 raw = adc1_get_raw(adc1_channel);
      |                       ^~~~~~~~~~~~
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:588:44: error: 'ADC_WIDTH_12Bit' was not declared in this scope; did you mean 'ADC_WIDTH_MAX'?
  588 |                 adc2_get_raw(adc2_channel, ADC_WIDTH_12Bit, &raw);
      |                                            ^~~~~~~~~~~~~~~
      |                                            ADC_WIDTH_MAX
/Users/fl/Documents/Arduino/libraries/ESP32AnalogRead/ESP32AnalogRead.cpp:588:17: error: 'adc2_get_raw' was not declared in this scope
  588 |                 adc2_get_raw(adc2_channel, ADC_WIDTH_12Bit, &raw);
      |                 ^~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

garthpaine, 2024-06-16:
I also have the same issue. any ideas welcome.

purehunter1, 2024-07-02:
I have similar compiler errors since updating the Arduino-ESP32-core to version 3.x

Main problem seems to be the deprecated ESP library "esp_adc_cal.h" which is used by this lib (ESPAnalogRead.h)
`#warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"`
...
`error: 'adc1_config_width' was not declared in this scope
`
...
and so on

ybpvin, 2024-07-03:
> I have similar compiler errors since updating the Arduino-ESP32-core to version 3.x
>
> Main problem seems to be the deprecated ESP library "esp_adc_cal.h" which is used by this lib (ESPAnalogRead.h) [...]

Same! I reverted to an older version of the kernel. In the future, I will probably look for a new library

matthias-bs, 2024-07-03:
The Arduino-esp32-core now provides a function `analogReadMilliVolts()` which applies the calibration - see https://docs.espressif.com/projects/arduino-esp32/en/latest/api/adc.html#analogreadmillivolts.

So it seems `ESP32AnalogRead` (or a similar third party library) is no longer needed.

`ESP32AnalogRead` filled the gap for quite a while and did its job perfectly fine. @madhephaestus Thanks again!

ybpvin, 2024-07-04:
> The Arduino-esp32-core now provides a function `analogReadMilliVolts()` which applies the calibration - see [...]

Thanks for the answer and advice!